ExpressLRS/Backpack #28: TBS Fusion VRX integration

microraptor, 2021-11-04:
The TBS Fusion VRX module can connect to the WiFi of a TBS Crossfire/Tracer module, which enables these features:

- automatic VTX/VRX frequency sync
- dynamic VTX power
- telemetry OSD inserted on the VRX side

See page 24 onwards at: https://www.team-blacksheep.com/tbs-fusion-manual.pdf

It would be great, if someone could figure what kind of data/protocol the TBS Fusion needs for this connection. So that it will be possible to connect a TBS Fusion to an ExpressLRS TX with an ESP backpack and use these features.

@JyeSmith has already got the frequency sync working with a DIY VRX module: https://github.com/ExpressLRS/ExpressLRS/wiki/VRx-Backpack

In an earlier issue it is mentioned, that apparently [TBS Fusion integration is already being worked on](https://github.com/ExpressLRS/ExpressLRS/issues/839#issuecomment-961178500).

JyeSmith, 2021-11-05:
If you have the tx BP and latest Configurator, you can start to test this with https://github.com/ExpressLRS/Backpack/tree/add-fusion-target

Jump on Discord for flashing support etc

microraptor, 2021-11-05:
Thank you for your work! Great that it even uses the builtin ESP. I will try it when I have some time. Is the dynamic VTX power feature planned for the future or does it seem too difficult to implement it?

JyeSmith, 2021-11-05:
No immediate plans.

JyeSmith, 2022-01-27:
Target merged.

closed 2022-01-27 by JyeSmith