ExpressLRS/Backpack #38: The firmware for HappyModel EP82 VRX-Backpack module

L1cardo, 2021-12-22:


JyeSmith, 2021-12-22:
For now if you hook it up and flash the RX target it will work fine.

I hooked power up to mine, press the button to enter WiFi mode, then uploaded the binary.

https://github.com/ExpressLRS/Backpack/wiki/Using-an-ESP-based-receiver

jklong03, 2021-12-22:
> For now if you hook it up and flash the RX target it will work fine.
> 
> I hooked power up to mine, press the button to enter WiFi mode, then uploaded the binary.
> 
> https://github.com/ExpressLRS/Backpack/wiki/Using-an-ESP-based-receiver

To avoid any confusion, shouldn't this be clarified to "flash the VRx target for the Happymodel EP(x) 2.4ghz family from the Backpack Tab in the new Configurator" or similar verbiage ?

techfixpros, 2021-12-28:
Please let me know if this needs to be its own thread

RapidFire+: 1.3.2
HM ES900Tx: ELRS 2.0
HM EP82: Tried both 0.2.0 Rapidfire ESP Rx & 0.2.0 RapidFire ESP01F
LUA: v2.0
EdgeTx: 2.5.0 on Original Taranis X9D(not +)

Flashed RapidFire ESP Rx 0.2.0, soldered into module, and the RF+ module works, ESP82 will not do anything other than solid blue LED. Button does not change to WiFi, and power-cycling goggles does not make EP82 enter bind mode.

Flashed RapidFire ESP01F back onto EP82 and installed in module, RF module works fine, but the EP82 is completely unresponsive. It will not enter bind mode after power-cycling goggles, and button does not make it go into WiFi mode anymore like it did when the module was disconnected from the module.

Flashed the RF ESP01F before installing in the RF+ module, and the button will let me go into bind, WiFi, and run modes. Binding does not actually bind and the LUA does not make the EP82 go into WiFi mode.
[image]

L1cardo, 2021-12-28:
I flashed it using ELRS Configurator with a binding phrase, and it works fine, without asking me to manually bind. Maybe you can try to build your own firmware with a binding phrase.
Also, make sure your wires are soldered correctly.

techfixpros, 2021-12-28:
> I flashed ESP01F using ELRS Configurator with a binding phrase, and it works fine, without asking me to manually bind. Maybe you can try to build your own firmware with a binding phrase.
> Also, make sure your wires are soldered correctly.

Tried that too, it flashed but no luck.
Question, are you using the EP82? And if so, did you use the discrete M0, Clk, Cs pads, or follow the wiki and solder to Tx, Rx, and whichever is the boot pad?

Aaronsss, 2022-01-08:
@techfixpros did you just use an ftdi to program the esp82 to 0.2.0?

I have a similar issue with my HM esp82 backpack. Basically the first time I connected it to the goggles the rapidfire said spi mode on the screen. After power cycling it wouldn't do anything or let me bind it was just a solid blue light.

I removed from the RF and programmed it with an ftdi programmer. That was successful and I could command it to WiFi mode from the controller whilst powered from the ftdi

After hooking back up to the module I'm back to the same solid blue light and not doing anything again

techfixpros, 2022-01-08:
Found the (chinese?) youtube from mfg and it stated using the ESP01F RapidFire firmware so I flashed that via FTDI to get it off the non-functional esp version. Then cleaned with isopropyl and made sure there were no solder bridges or stray wire strands. I think my issue was a short caused by my "no-clean" flux not being as no-clean as it should have been because without changing wiring, it worked properly afterwards. 
I used the configurator to hard code the WiFi and binding phrase. Ohmed out the mCU pins to the pads and verified you need the M0, SCK, and SCL pads per the /targets/diy_rx.ini, and not the Tx/Rx pads.
Also had to flash the Tx backpack to the Tx 0.2.0 per deadbyte on Discord.

That's all I know that fixed mine. It consistently switches the VRx to SPI mode even after numerous power-cycles, and changes band/freq without issue now.

Aaronsss, 2022-01-08:
I have discovered what my issue is. HDO's are messing about with the SPI lines and preventing it working properly. Time to cut the pins off!

techfixpros, 2022-01-08:
That's what I have read, in my case it works fine in my HDO2 with the pins intact. Sad that the buttons no longer work though.

Aaronsss, 2022-01-08:
Who uses those buttons on the fatsharks anyways :) but I can confirm it works now. May just be a HDO thing

closed 2022-06-11 by pkendall64

HazzaHFPV, 2023-02-15:
My Happymodel EP82 and my Rapidfire are at the moment not working together either. The EP82 fails to put the rapidfire into SPI mode.

What I noticed is: when I make sure that the rapidfire fully boots up first and then connect the EP82 to power, it works fine. If the EP82 boots first, then the rapidfire will not enter SPI mode.

So my current problem is, when I connect them to power at the same time, the rapidfire finishes its boot sequence not in time and the EP82 then fails to switch it into SPI mode as the rapidfire is "late to the party"...?

I don't know how to solve that right now, as when they are all wired up and installed in the goggles, I have no control over the sequence. I am also surpised that there are so many tutorials about the EP82 and everyones seems to work fine..

Firmware on the EP82 is the Rapidfire_HappyModel_EP82_VRX_Backpack-1.2.0
Firmware on the Rapidfire is version 1.3.3.

pkendall64, 2023-02-15:
> My Happymodel EP82 and my Rapidfire are at the moment not working together either. The EP82 fails to put the rapidfire into SPI mode.
> 

Could you flash the master branch and test please as we have a fix in place for that.

HazzaHFPV, 2023-02-15:
Yes! Master branch is working :) Thank you, you saved me a lot of fiddling around.
I don't know why I didn't think of trying the master-branch! Was too focused in I guess?
Thank you very much :)

pkendall64, 2023-02-15:
> Yes! Master branch is working :) Thank you, you saved me a lot of fiddling around. I don't know why I didn't think of trying the master-branch! Was too focused in I guess? Thank you very much :)

Thank you for testing and reporting back. 👍

techfixpros, 2023-02-19:
Issue is resolved, thank you.