ExpressLRS/Backpack #41: Graphics for possible wiki entry for Fusion flashing via FTDI

jklong03, 2022-01-03:
Here are a few graphic aides to help in developing the wiki entry for flashing the internal Fusion esp-12f module via FTDI .
I am providing them to help someone flesh-out the wiki entry if they desire to use them.
[image]
[image]

sendorm, 2022-01-11:
I've used this method and found to be easier than holding pins to ground. 
[image]
After making the connections: Hold the joystick to the left while powering the Fusion module by USB, during flashing via configurator. 
https://team-blacksheep.freshdesk.com/support/solutions/articles/4000153255-tbs-cloud-unbrick-the-wifi-module

jklong03, 2022-01-12:
While I too have used this one, the advantage of the simple grounding of GPIO0 and briefly grounding the RST pin is that the latter method works after power up. It doesn't require the timing of holding joystick and plugging in usb on boot, works anytime after Fusion is up. Personal preference I guess.

sendorm, 2022-01-12:
I didn't like/manage trying to touch a miniature pad while trying to hold another one at the same, with the unknown stress of what happens if I touch something else. In the end just solidly soldering 2 pins and pressing a joystick was easier for me. And also found out that the timing was quite forgiving. Also you use the official configurator. Which will be easier for most.
But I agree, it all comes down to preference.

jklong03, 2022-01-13:
Just a quick FYI, I usually have been building in VSC and flashing with the
esp8266 flasher standalone.

Jeffrey Long
Cap'nFPV
Captain, USAF, Ret.

L1cardo, 2023-01-31:
[image]