ExpressLRS/Backpack #42: Ability to disable backpack / channel change on start up

Aaronsss, 2022-01-08:
Not all of my models run on ELRS so it is annoying that it will change the channel to R8 when you turn the goggles on.

Is it possible to add a feature that prevents this change if the bound ELRS modules is not available?

jklong03, 2022-01-08:
What VRx module are you using? Most of the wiring diagrams need the CS pin removed, which should prevent this. On Fusion Module you can also designate a default startup channel by saving it as  favorite channel and then long press to save as startup.

Aaronsss, 2022-01-08:
I am using the rapid fire. So I need to disconnect the cs pin to prevent it controlling the vrx? There must be a nice way to do that in the firmware

L1cardo, 2022-01-13:
Yeah, I need this too. I am using rapid fire as well, and I don't want it to be R8 every time I turn it on.

PR #44 (2022-01-23): Don't init the VRX until the first VRX command
This makes the module stay on the last know channel if the radio is off instead of changing to R8 (rapidfire)

If the radio is on or a lua command is sent it works as before.

This closes #42 

Edit: thinking about it this may not work well for other vrx modules. I only have the rapid fire to test with. Is it worth moving this into rapidfire.cpp?
closed 2022-01-23 by JyeSmith