ExpressLRS/Backpack #43: Eachine pro58

Moksh-2000, 2022-01-09:
Any possibility for eachine pro58 ?
It's a nice vrx at low price work like charm as nearly fusion compared

Nordle420, 2022-03-16:
yes please 🥺

jonas-koeritz, 2022-06-21:
It seems like there are many different firmwares available for this module. Is there one specific one you would like to see support for? Maybe the developers of that firmware can provide documentation on how to control the module through SPI or something similar.

techfixpros, 2022-10-01:
> It seems like there are many different firmwares available for this module. Is there one specific one you would like to see support for? Maybe the developers of that firmware can provide documentation on how to control the module through SPI or something similar.

Achilles would probably be the most ideal and firmware to add this to. It is still being actively developed on.

DAMskill, 2025-01-08:
I'll receive the Eachine Pro58 soon. Is there any progression in the ELRS Backpack functionality feature?

ppyzik, 2025-07-26:
Form what I have discovered for now during my research -  the only open source firmware I have found that is compatible with PRO58 is https://github.com/piodabro/WPro58

This is project is based on https://github.com/sheaivey/rx5808-pro-diversity

Other project that based on rx5808-pro-diversity is FENIX that is supported / supports ELRS backpack: 
https://github.com/JyeSmith/FENIX-rx5808-pro-diversity/

Conclusion is that it highly possible to integrate FENIX features into WPro58 to make it work with ELRS bacpack