ExpressLRS/Backpack #46: Backpack as connection to Antenna tracker?!

Pairan, 2022-01-18:
That was the first that came up in mind as I read the description:

* Place an antenna tracker (e.g. MFD Crossbow mini) and attach an esp-device to its port
* bind it to the backpack and transmit the telemetry to it so the tracker can work

would this be possible?

TwoToneEddy, 2022-04-21:
This sounds like the ideal application for backpack! I was thinking the same.

Pairan, 2022-04-23:
correct ... I assume ESPnow to be a good link to the RX-Backpack and deliver the data to the Tracker as it requires it

CorruptName, 2023-01-16:
Has anyone figured out a solution for this?

Pairan, 2023-01-17:
not yet ... but I assume that should be doable. I'm just not that good at backpack / C and would need someone to do it for us

SkylarTheGrey, 2023-01-28:
I cant help code it but I can comment that this would be a excellent feature for several people I fly with and myself!

pacaj2am, 2024-06-04:
should be closed with this pull request: https://github.com/ExpressLRS/Backpack/pull/126

closed 2024-06-04 by pkendall64