ExpressLRS/Backpack #55: [Feature Request]

nightflasher, 2022-01-27:
I don't know, who did it first, but there was a communication "proof of concept" with the lilygo T-Watch-2020. Maybe the backpack could communicate with the T-Watch to send telemetry data, and the module can be configured via the watch?

JyeSmith, 2022-01-27:
Peho did this.  Shouldn't be a problem to do, but does require some time and a killer use case.

nightflasher, 2022-02-03:
Every enhancement is a killer use case.
I think about getting telemetry on this device and setting alarms for battery, rssi and of course configuring the module - this comes in handy, when using transmitters that doesn't have Lua or modules without config screen/button.

pkendall64, 2025-01-03:
PRs welcome 😄

closed 2025-01-03 by pkendall64