ExpressLRS/Backpack #63: Namimno Flash OLED TX backpack firmware

L1cardo, 2022-03-10:
So, Namimno Flash OLED TX uses HappyModel backpack firmware?
Even if the target and the code are exactly the same, can we make it more clear when naming or displaying? 

[image]
[image]

miggy1952, 2022-05-03:
I also see that it is flashing happy model firmware onto my Naminno OLED hardware

Schnuppi12, 2022-08-26:
Here the same. And my Namimno OLED TX is complaining, that I want to flash the wrong firmware. 
Maybe it can be cleared up for people, that they will not be confused in the future, when they are updating their backpack.

LOFT330, 2023-01-16:
I am currently having the same issue and when I force the update it will fail. Has anyone found the fix?

Lukanex, 2023-08-16:
The update never ends in my case.
[image]
Any ideas please? 

I tried the update over WiFi (both onboard wifi connected to my network, each with a different IP address, so I chose the one of the backpack)
and also with UART.

Lukanex, 2023-08-17:
Hi I would like to share a challenge and a solution I found for it if anyone searches for the same:
--------------------------------------------------------------------------------
I am getting the following error when trying to reflash my backpack:
```
Processing HappyModel_TX_Backpack_via_WIFI (board: esp8285; platform: espressif8266@4.2.0; framework: arduino)
--------------------------------------------------------------------------------
UnknownPlatform: Unknown development platform 'PackageItem <path=%user%\.platformio\platforms\espressif8266 metadata=PackageMetadata <type=platform name=espressif8266 version=4.2.0 spec={'owner': 'platformio', 'id': 8008, 'name': 'espressif8266', 'requirements': None, 'uri': None}'
```

What's even stranger, this warning is displayed after the "flashing device" was indicated.
[image]

This happens in both cases whether I choose BUILD or FLASH.
It fails at the building or flashing time and complains that either buid or flashing failed.

[image]

--------------------------------------------------------------------------------
I found a solution in this post:
https://github.com/platformio/platformio-home/issues/1964#issuecomment-1086030600
`Try to remove C:\Users\blusc\.platformio\platforms\espressif8266 folder.`
--------------------------------------------------------------------------------
But that is not the whole story. The Configurator then downloads the package, upacks it and is stuck there forever (>20 min):
[image]

pkendall64, 2025-01-03:
Old and closing as this should be resolved with the alias targets

closed 2025-01-03 by pkendall64