ExpressLRS/Backpack #68: wrong link?

psicho100pro, 2022-05-12:
Hi to all,
https://github.com/ExpressLRS/Backpack/wiki/Wiring-Guides#generic-rx5808-module
Here is the link that leads to the creation of a new page.
"https://github.com/ExpressLRS/Backpack/wiki/RX5808-Backpack-using-an-ELRS-receiver"
I don't know if it's a mistake or waiting for someone to take on the role of guide.
I was just looking for some information I found elsewhere and came across this.
Tribute to your amazing work.

JyeSmith, 2022-05-12:
This is my fault.  I just never got around to writing the page :|

lesharodin, 2023-01-20:
> This is my fault. I just never got around to writing the page :|
[image]

Add this image to it. It works, tested yesterday to convert my Boscam FPV watch to ELRS Smart FPV Watch

lesharodin, 2023-03-15:
@pkendall64 @JyeSmith  is backback repo abandoned here? :(

pkendall64, 2023-03-15:
> @pkendall64 @JyeSmith is backback repo abandoned here? :(

No it is not abandoned