ExpressLRS/Backpack #74: Rapidfire + HM EP82 on v1.2.0 does not enter SPI mode

techfixpros, 2023-01-01:
Rapidfire 1.3.2 (also tested and fails on 1.3.3)
HappyModel EP82 1.2.0 (works fine on 1.1.0)

Tx can control EP82 on both 1.2.0 and 1.1.0 fine.
Rapidfire module does not enter SPI mode when on 1.2.0 per LCD screen.
When channel/band is changed on Tx, Rapidfire will jump through different channel and band settings for about 2 seconds and stop, but not change to match Tx commanded setting. Multiple power-cycles does not correct operation.
Flashed back to 1.1.0 and issue is resolved with SPI mode shown on screen, and RF changing immediately to commanded setting.

mattiachiarelli, 2023-01-28:
I had the same issue... Tires flashing 1.1 and now it works

pkendall64, 2023-02-15:
> I had the same issue... Tires flashing 1.1 and now it works
> 
> I'll add that clicking the ep82 button makes the rapidfire go to spi (2 times because you enter and than exit wifi mode)

Could you try with the master branch of the backpack and report back, as we've merged a fix for this.

pkendall64, 2023-02-15:
@techfixpros @mattiachiarelli another user has reported that the master branch does indeed fix this issue with SPI on rapidFIRE.

mattiachiarelli, 2023-02-15:
> .

Hi! Thank you!! I will test it this weekend

techfixpros, 2023-02-15:
I will try it this weekend.

techfixpros, 2023-02-19:
Confirmed issue is resolved. Thank you.

closed 2023-02-19 by techfixpros