ExpressLRS/Backpack #79: 3 time blink?

lesharodin, 2023-01-19:
Hi, I have a strange issue:
Backpack works fine, but when telemetry is on and quad is connected VRX backpack blinks 3 times than pause.
Disconnect quad - solid LED on VRX Backpack.
When it blinks the FPV screen blinks too. (If I disconnect data pin it stops, so its transmitting smth).
When I turn telemetry OFF in LUA scrips it stop blinking and works fine.
I tried to use different binding phrase on backpack, but issue is not there.

2.5.1 ELRS on TX
1.2 Backpack

pkendall64, 2023-01-19:
This is normal behaviour. When the TX module detects a connection to an RX (i.e. it receives telemetry) it waits for a second or so for the FC to become ready, then it sends out the commands to change VTX channel, to both the VRx and the RX for forwarding to the FC.

lesharodin, 2023-01-20:
> This is normal behaviour. When the TX module detects a connection to an RX (i.e. it receives telemetry) it waits for a second or so for the FC to become ready, then it sends out the commands to change VTX channel, to both the VRx and the RX for forwarding to the FC.

It was doing it continiously. Every 1-2 sec or so. It must do it only when I press "send vtx"?
Strange enough after I swith telem off and then on it starts working normally..

closed 2024-11-17 by pkendall64