ExpressLRS/Backpack #80: Happymodel EP82 target aliasses.

lesharodin, 2023-01-20:
Problem:
We only have rapidfire HappyModel EP82 target, but it doesnt even say that its for rapidfire only in configurator:
[image]
I was strugling whole evening with that, than I dig in to repo and check the target itself (https://github.com/ExpressLRS/Backpack/blob/master/targets/happymodel_vrx.ini) and find out that its rapidfire only.
From user perspective its imposible to find out that HappyModel target is for rapidfire, and that EP82 is same as ESP01F. 

Tried to make PR https://github.com/ExpressLRS/Backpack/pull/78 and find out that configurator doens't show new target from PR selected.

We need to add aliases for all ESP01F for HappyModel EP82, or just make documentation for backpack better. (Can anyone told me how can I contribute for wiki?)

jurgelenas, 2023-05-25:
Hey @lesharodin, 

>  I was strugling whole evening with that, than I dig in to repo and check the target itself (https://github.com/ExpressLRS/Backpack/blob/master/targets/happymodel_vrx.ini) and find out that its rapidfire only.

I have renamed the target in https://github.com/ExpressLRS/ExpressLRS-Configurator/commit/f1db3ff17e9500f4438673f0cc515b3156c18cc2. 

> Tried to make PR https://github.com/ExpressLRS/Backpack/pull/78 and find out that configurator doens't show new target from PR selected.

You need a separate PR for new targets so the Configurator app is able to display these. You can see many examples in [closed pull requests](https://github.com/ExpressLRS/ExpressLRS-Configurator/pulls?q=is%3Apr+is%3Aclosed) on the configurator repository. One of the examples: https://github.com/ExpressLRS/ExpressLRS-Configurator/pull/491

> We need to add aliases for all ESP01F for HappyModel EP82, or just make documentation for backpack better. (Can anyone told me how can I contribute for wiki?)

ExpressLRS.org is developed on the Docs repository: https://github.com/ExpressLRS/Docs. For contributions it would be best to get into the contact with @deadbytefpv on our discord server to discuss.

pkendall64, 2025-01-03:
I believe this has been addressed with the addition of extra targets

closed 2025-01-03 by pkendall64