ExpressLRS/Backpack #93: Can't override video channel with Skyzone Steadyview+ELRS receiver

odddollar, 2023-06-19:
I'm using the [Skyzone Steadyview with built in ELRS backpack board](https://www.skyzonefpv.com/en-au/collections/goggle-parts/products/steadyview-elrs-backpack-receiver-for-fpv-goggles-48ch-5-8ghz-v3-3-hardware) as the receiver in my goggles (Skyzone Cobra S). I have it bound to my transmitter and can change both the VTX and VRX channels successfully through the ELRS VTX administrator page of the LUA script. However, I am not able manually override the VRX channel through the UI of the goggles, meaning I am only able to change VRX channels through my transmitter when I have an ELRS model selected.

Not all of my models with video transmitters use ELRS, and when I'm watching another person's video I won't have my transmitter on and thus won't be able to change my VRX channel.

Is there a way the ELRS backpack could set the VRX channel initially, but permit overriding through the goggle UI to manually switch to other channels? If not then my video receiver is practically unusable and I would've been better off buying the non-ELRS backpack version of the Steadyview module.

pkendall64, 2023-06-19:
Nope, not possible.You had to remove the pins so the goggles can no longer change the channel on the Steadyview.
The problem with leaving the pins in the module is that the goggles and the backpack drive the pins and so neither will work.

closed 2023-06-19 by pkendall64

odddollar, 2023-06-19:
But the backpack does work and I haven't removed the pins. If this is working as intended this is very poor design for a VRX module with no external way of controlling the channel. Not all of my models use ELRS so you're saying I have to change channel through my transmitter? If this is the case then I'm going to return this module and get one without the backpack.

odddollar, 2023-06-19:
@pkendall64

pkendall64, 2023-06-19:
Ah, sorry I misread your initial description. You have the "Skyzone Steadyview with ELRS backpack" not the DIY backpack like I thought. That should definitely allow changing via the goggles or via the backpack.

odddollar, 2023-06-19:
Sorry, I posted another issue with updated clarifications because I knew it wasn't working properly and I didn't think anyone would see this one.

I've contacted Skyzone support as I think it might be a firmware issue with my goggles (that I can't manually change channel), but I can't say I have high hopes that it'll be fixed.