ExpressLRS/Backpack #102: Flashing ExpressLRS Backpack 1.3.0 to HappyModel EP82 "Update error: ERROR[4]: Not Enough Space"

slash128v6, 2023-08-17:
I have an ImmersionRC Rapidfire analog module with a HappyModel EP82 VRX backpack. Building the 1.3.0 backpack firmware using ExpressLRS configurator 1.6.0 for VRX target HappyModel EP82 VRX Rapidfire Backpack for WiFi flash method results in file "Rapidfire_HappyModel_EP82_VRX_Backpack-1.3.0.bin". When attempting to flash this to the EP82 via WiFi I get "Update error: ERROR[4]: Not Enough Space".

slash128v6, 2023-08-31:
I was able to work around this by flashing 1.2.0 first. But I had to use PlatformIO and VSCode to build because the hdzero.ini file had dots (.) in the env name which caused ExpressLRS configurator to throw errors. After building I flashed 1.2.0 to the EP82 and then was able to build and flash 1.3.0 to the EP82.

pacaj2am, 2024-06-05:
had same issue with flashing 1.4.1. needed to reflash to be sure, what is the binding phrase. Flashing 1.3.0 and than 1.4.1 back helped

pkendall64, 2025-01-03:
The configurator and we-flasher now provide gzipped which can be uploaded and avoids this error

closed 2025-01-03 by pkendall64