ExpressLRS/Backpack #104: VRX set F8 when select R7

aQuaplug, 2023-09-06:
Hi, I'm using TBS Fusion with internal backpack flashed.
Thanks to it works very well.
But one annoying thing is VRX switched to F8(5880) when I select R7(5880) with LUA.
Yes it's same frequency but little annoying when I switch manually with module's button for watching other pilot's raceband channel.
And just want to see R7 on Fusion's OLED :)

What did I miss? or just intended?

pkendall64, 2023-09-06:
It's just an artefact of the way the Fusion works. The backpack gets an index into the standard VTX table, in the fusion backpack code, it looks up the index to get the frequency and sends that to the Fusion module. The Fusion module then displays what it wants to given the frequency.
So, I'm afraid theres not a lot we can do about this one.

closed 2023-09-06 by pkendall64

aQuaplug, 2023-09-06:
Oh, I got it. thanks :)