ExpressLRS/Backpack #118: Backpack repo missing binary_configurator.py

i-am-grub, 2023-11-24:
Currently unable to the backpack over UART through the configurator due to the backpack repo not having a `binary_configurator.py` file. 

While most devices likely are flashed over wifi, it is useful to have the UART option for flashing DIY devices.

pkendall64, 2023-11-25:
The backpack repo does not need binary_configurator as the binary_flash program does everything that is required.

closed 2023-11-25 by pkendall64

men1al, 2024-04-27:
Hi,
I have the same issue, when trying to build the backpack firmware with the ELRS configurator following instructions "Installing the timer's backpack" from here : https://github.com/i-am-grub/VRxC_ELRS/releases/tag/v1.0.0

I get a Build error "\\AppData\\Roaming\\ExpressLRS Configurator\\firmwares\\binary\\Backpack\\python\\binary_configurator.py': [Errno 2] No such file or directory".
It happens with any settings in Git Pull Request tab.

- ExpressLRS Configurator v1.6.1. Reinstalled and cleared platformio dependencies.
- Windows 11 Pro, tested with MacOS with same error
- Settings: Generic Race Timer Backpack #114 / RotorHazard / ESP32 / WIFI / Build
Same issue with different settings or UART.
- There is no issue building from Backpack/Official Releases

Has someone an idea ? :)
Thanks !