ExpressLRS/Backpack #127: TBS Fusion Backpack not working

robschii, 2023-12-30:
Firware Version
Backpack 1.4.1 on RM Boxer with last ELRS internal
FUSION Firmware 2.22  (other also testet till 2.10)

Firmware was falshed tu fusion with fdi with bindig phrase and homenetwork

When i am in elrs lua and will bind again the fusion esp is blinking one time, so it is communicate with the RM Boxer.
The WifiSymbol is with an "X" sign (not connected). Fusion is set in vtx follow mode, wifi is on

In VTX Admin i change the channel and push sent to vtx. VTX is changed but nor the fusion.

i have no idea anymore whats wrong.

i trry it the first time 3afte sold my tango 2 for elrs

robschii, 2024-01-03:
got it, i flashed 1.3.0 and it works , when i updated it to 1.4.1 via wifi its brick again. binding works (led on esp is blinking) but no changing channel or to bring the module to wifi ap  is also not working.  back to 1.3.0 via fdti its working again.   fun fact the radio is on 1.4.1
[11:03]
bug is that r7 is shown F8  but its the same frequenzy

f4b0, 2024-01-06:
Hi, i can confirm that, have the same problem here with my TBS Fusion. 
After update to 1.4.1 its not talking with the transmitter any more and i can not change the channels or enable the wifi.. .

robschii, 2024-01-07:
i was also switch it on and off but ist was not working with 1.4.1

pkendall64, 2025-01-03:
There was a bug in 1.4.1 to do with the flashed bind-phrase not taking effect and having to bind from the TX module.
This was fixed in 1.5.0

closed 2025-01-03 by pkendall64