ExpressLRS/Backpack #129: Jumper t20 gemini

KondratokGK, 2024-01-10:
Is the Jumper T20 gemini supported?

pkendall64, 2024-01-11:
You can flash it with the generic and then ESP32 based.

KondratokGK, 2024-01-11:
As far as i know it uses additional ESP8285 for espnow, right?

pkendall64, 2024-01-11:
Yes, the code runs on the ESP8285 (backpack) but talks to the ESP32 (main ELRS MCU).
Select the following options in the backpack firmware upload section of configurator.
[image]

KondratokGK, 2024-01-11:
Ok. If I understand correctly, it should look like this: [Jumper T20] <-CSRF-> [Internal ESP32 RX] <-UART-> [?additional ESP8285 board?] <- ESP Now-> [Steadyview ELRS VRX].

closed 2024-07-27 by pkendall64