ExpressLRS/Backpack #139: headtracker

ltwin8, 2024-06-28:
Is there any documentation on a DIY headtracker available? which imu is needed? would it work on an EP82 backpack with external imu (super simple and quick) or is an ESP32 like in the HDZero goggles nesessary?

pkendall64, 2024-06-28:
The current implementation in the backpack just allow the goggles to forward head-tracking information back the to handset module. There is currently no IMU code in the backpack. However we are working on a branch that does provide this functionality. When that is ready there will be information available, but this project is on the back-burner a little bit.

ltwin8, 2024-06-30:
is there anything i can pick up on? i asked on discord already but no helpfull answer?

it seems the basics are already implemented, transmitting headtracking to the backpack.... i do not see what would be the big problem in using it stand alone?

or is there any timeframe on when it will be officially available?

closed 2024-11-17 by pkendall64