ExpressLRS/Backpack #157: Retain .bin extension for gziped binaries

jurgelenas, 2024-08-20:
This PR https://github.com/ExpressLRS/Backpack/pull/156 has introduced an inconsistent behavior when compared to the main firmware repo by producing `.bin.gz` filenames for WiFi builds. We need to maintain exactly the same behavior as the main firmware.

pkendall64, 2024-08-21:
Not sure I understand the issue! Both backpack builds and main firmware build produce a folder with just a firmware.bin.gz file in them for 8285 based targets.

jurgelenas, 2024-08-21:
> Not sure I understand the issue! Both backpack builds and main firmware build produce a folder with just a firmware.bin.gz file in them for 8285 based targets.

Yeah my bad on this one! After further testing everything looks good.

closed 2024-08-21 by jurgelenas

pkendall64, 2024-08-21:
> Yeah my bad on this one! After further testing everything looks good.

Whew, I thought I stuffed something up!