ExpressLRS/Backpack #161: Compiling backpack firmware with 1.7.6 flatpak is missing pyserial

froqstar, 2024-09-19:
When trying to compile the current backpack headtracker PR using configurator 1.7.6 flatpak (from flathub), it fails at the binary configuration step with 
```
File "/home/froqstar/.var/app/org.expresslrs.ExpressLRSConfigurator/config/ExpressLRS Configurator/firmwares/binary/Backpack/python/serials_find.py", line 1, in <module>

    import serial
ModuleNotFoundError: No module named 'serial'
```
Full log:
[ExpressLRSBuildLog_20240919183518664.txt](https://github.com/user-attachments/files/17064651/ExpressLRSBuildLog_20240919183518664.txt)

pkendall64, 2024-09-19:
Seems a bit weird that configurator hasn't installed pyserial!
You can manually install it with pip globally to get this working.

froqstar, 2024-09-19:
Thanks, I tried that but it isn't working. Maybe because the host OS is using python 3.13 but the flatpak runtime is using 3.11 (I think, it's missing from the attached logs but says so at the start of the compilation step).