ExpressLRS/Backpack #169: Skyzone Skyzone steadyview X

mishakiev, 2024-11-07:
Skyzone steadyview X not working with external diy backpack.

pkendall64, 2024-11-07:
The Steadyview X is not supported via the DIY backpack as it does not use SPI communication. It's uses a UART/serial communications.

closed 2024-11-07 by pkendall64

mishakiev, 2024-11-08:
Is the protocol known?

pkendall64, 2024-11-08:
> Is the protocol known?

Why yes it is 😄. See <https://github.com/pkendall64/hdzero-goggle/blob/analog-module-control/src/module/steadyview_x.c> where the serial connection is 115200, N, 8, 1

shatohingv, 2025-01-23:
> > Is the protocol known?
> 
> Why yes it is 😄. See https://github.com/pkendall64/hdzero-goggle/blob/analog-module-control/src/module/steadyview_x.c where the serial connection is 115200, N, 8, 1

what are the pins for connecting uart to steadyview x??

pkendall64, 2025-01-24:
Heres a document that I created when I was reverse engineering the protocol.
[Steadyview X.pdf](https://github.com/user-attachments/files/18528467/Steadyview.X.pdf)

shatohingv, 2025-01-31:
> [@shatohingv](https://github.com/shatohingv) Heres a document that I created when I was reverse engineering the protocol. [Steadyview X.pdf](https://github.com/user-attachments/files/18528467/Steadyview.X.pdf)

thanks

TieNotTie, 2025-02-06:
@pkendall64 , could you please clarify how to find TX and RX pins on SteadyViewX described in the document you have attached? I can see type-c port and 9 pins but which of them are 'Module RX' and 'Module TX'?

pkendall64, 2025-02-06:
Pins 1, 2, and 3 of the module bay connector.
1 - module RX
2 - module TX
3 - pull low

TieNotTie, 2025-02-09:
Thank you for your answer. But I am wondering whether is it possible somehow to control SteadyView X, using it's built-in pins. I do not have hdzero goggles, so I cannot try it by myself. But I have previously tried to replace a built-in steadyview module in skyzone goggles and it worked fine. I assume they have some kind similar control interface but I couldnt manage it. I am able to control skyzone steadyview module without goggles, using my own hardware by SPI protocol. Maybe you have any ideas how to change channels/band on SteadyView X or how it's communication interface differs from Skyzone SteadyView?