ExpressLRS/Backpack #178: Trying backpack on the Speedybee VRX with an adapter

flyasky, 2025-03-02:
Connected as below:

pkendall64, 2025-03-02:
Are you sure the speedybee VRx is controllable using SPI via those pins? Usually they are only for button control. i.e. can only select 1 of 8 different channels.

flyasky, 2025-03-02:
You mean, it could react to pull to +5V as a joystick button clicks?
Unfortunately, I can't find any protocol instructions for these pins for this speedybee vrx.
I also tested with other firmware (orca, rapidFire) and (ep82, esp01s) for some of them I see that the channel menu is opened and something is going on, but the result is unpredictible.

flyasky, 2025-03-02:
I've seen this tread made me think it does support SPI: https://www.reddit.com/r/fpv/comments/1chqv86/speedybee_58ghz_receiver_pinout/

pkendall64, 2025-03-02:
The pinout is the same on all modules, but the feature set it supports is differrent.
You will need to contact speedybee and see if they support SPI protocol on the first 3 pins or if it's just for the old-style fatshark buttons. u.e. a 3 bit value for the channel number.

The Rapidfire supports button mode and SPI, the Steadyview supports SPI, the SteadyviewX supports UART and the TBS Fusion only supports button mode. So different modules have different levels of support.

blafpap, 2025-05-18:
@pkendall64 how do you know SteadyviewX use UART and steadyview SPI ? Curently i have a SteadyviewX and thought is will be the same as SteadyView.

pkendall64, 2025-05-18:
> [@pkendall64](https://github.com/pkendall64) how do you know SteadyviewX use UART and steadyview SPI ? Curently i have a SteadyviewX and thought is will be the same as SteadyView.

Because I traced them with a logic analyzer!

blafpap, 2025-05-18:
@pkendall64 Do you know where to search or do you have some info, If I want to change channels through uart control. how to wire the steadyviewX and how to talk with it?

pkendall64, 2025-05-18:
See #169 for details. I attached a PDF of my findings there